uf-mil-archive/PropaGator #10: Azi_Drive: Thrust clipping

jpanikulam, 2015-05-12:
Azi drive is auto-clipping incoming wrench commands to avoid impossible commands. 

When the vehicle is commanded with too large a wrench, azi-drive does not converge, and the existing azi-drive codebase does not handle this well.
This is half-solved, in the sense that the clipping will prevent a convergence problem in 99% of cases (Exceptions in the angle-ranging problem).

Lines causing this:
https://github.com/uf-mil/PropaGator/blob/master/azi_drive/nodes/azi_drive_node.py#L124

jpanikulam, 2015-05-14:
@zachgoins 

I won't have time to fix this until June.

Possible fixes:
1. Find the maximum wrench (at thruster saturation) for current configuration and set those to the clips

    min_T = B * [-100, -100].T
    max_T = B * [100,  100].T

2. Design a smarter controller/planner that commands in Newtons, N*m and knows to clip at realistic values.

fnivek, 2015-06-26:
Is this fixed? Would it be better to scale all values equally to maintain the direction of the force vector? Clipping would change the direction of the force vector which is bad for the controller.

commit 9b3f91e5 (2015-07-08): Azi_Drive: Speed up 25-50x, fix many bugs

Closes #9
Closes #10
Addresses #42

Note: This retains Fossen's nautical convention for boat coordinate
axes. You can put a negative sign in Y if it floats your boat.
sorry-not-sorry
commit 62da0b71 (2015-07-10): Azi_Drive: Speed up 25-50x, fix many bugs

Closes #9
Closes #10
Addresses #42

Note: This retains Fossen's nautical convention for boat coordinate
axes. You can put a negative sign in Y if it floats your boat.
sorry-not-sorry

Conflicts:
	azi_drive/nodes/azi_drive_node.py
	azi_drive/src/azimuth_drive/azimuth_drive.py
closed 2015-07-08 by jpanikulam