uf-mil-archive/PropaGator #11: Make sure thrust is always zeroed in all circumstances

fnivek, 2015-05-13:
I believe if the thruster_config.py node is shut down it does not zero the motors

jpanikulam, 2015-05-13:
One idea is to use services for setting thrust instead of a message stream. Services calls are still guaranteed to complete during a node's shutdown, [whereas messages are not](http://wiki.ros.org/rospy/Overview/Initialization%20and%20Shutdown#Registering_shutdown_hooks). Same is true in ROScpp afaik.

Azi Drive does this using a [control manager](https://github.com/uf-mil/PropaGator/blob/master/azi_drive/nodes/control_manager.py) node which provides a shutdown service. When you kill AD, it [calls a shutdown hook](https://github.com/uf-mil/PropaGator/blob/master/azi_drive/nodes/azi_drive_node.py#L107) and kills thrust using that service.

My thinking is to handle this natively inside of the thruster driver; because if the AD control manager crashes along with AD, then we are SOL.

fnivek, 2015-05-14:
So the control manager has to be killed after azi drive so they can't be in the same launch file? Otherwise the control manager could be killed before AD which would cause it not to stop the thrusters. Maybe you should make the control manager call stop_boat_service on shutdown... Except that uses msgs which are not guaranteed to be sent... hmm.

Well, we can work from lowest level up. Making sure that if anything in the chain breaks anything below it should be zeroed. But that would require a service at each level to make it safe, which is doable but seems like there is a better way. I guess I'll go through and convert all msg in the actuator chain into srvs instead.

P.S. ROScpp is the same but there is no shutdown hook you have to do it manually (not to bad though).

jpanikulam, 2015-05-14:
Yes, they control_manager and the rest of azi-drive cannot be in the same
launch file. However, services are guaranteed to be sent, so if mission
critical actions (thrust, servo angles) are set by service, then we can
change them in shutdown hooks. We just can't rely on messages.

Maybe adding timeouts to the motor controller is a good idea - "If I don't
receive a thrusterNewton command at least every 5 seconds, I will shut down
the motors"

fnivek, 2015-05-15:
I have an idea. The lowest level actuator control nodes should stop there actuators on shutdown and when a kill is issued from the global [kill handler](https://github.com/uf-mil/software-common/tree/master/kill_handling). Each node that updates actuators (including the lowest level nodes) (i.e. [tank steer](https://github.com/uf-mil/PropaGator/tree/master/tank_steer) or [thruster config](https://github.com/uf-mil/PropaGator/blob/master/motor_control/scripts/thruster_control.py))must send a kill to the kill service on shutdown. 

This method stops the actuators anytime the low level controllers are shutdown. Any time any node sends a Kill the actuators will stop. There is still an issue when the kill node gets stopped before any node that uses the global kill is shutdown. This could be solved a few different ways; 

Using services in the lowest level control nodes that the kill node would tell to stop when it is shutdown,

 by ensuring the kill node is always the first node to start and the last node to be shutdown,

or use the kill node as a heart beat, the low level nodes would have to receive not killed msgs from the kill node otherwise after some timeout they should stop there actuators. 

I'm going to investigate the final solution of the above three first.

fnivek, 2015-05-17:
About done with the upgrade! I'm updating the kill_handling to have the heartbeat built in. I'll have a readme of how to use it shortly.

fnivek, 2015-05-19:
That was a lot of work but its done! See the [PropaGator readme](https://github.com/uf-mil/PropaGator/blob/master/README.md) for the proper procedure to use actuators.

closed 2015-05-19 by fnivek