uf-mil-archive/PropaGator #23: Fix drift in Azi

fnivek, 2015-05-20:
You wrote this in my notes. I'm putting it in the issues to remind you.

jpanikulam, 2015-05-20:
Does this ref #9 ?

fnivek, 2015-05-21:
Is this the same issue Zach?

zachgoins, 2015-05-21:
Don't believe it is the same issue. The visualizer was showing it going straight but it still drifted a bit to the left.

jpanikulam, 2015-05-21:
The visualizer is using the raw B matrix assembled [here](https://github.com/uf-mil/PropaGator/blob/master/azi_drive/src/azimuth_drive/azimuth_drive.py#L97). 

- If it's drifting to the left when the visualizer thinks it is going straight, I *suspect* that is because the left and right thrusters are generating different thrust for the same value.

- If it is spinning when it supposed to be strafing, it is likely because the center of mass is offset from the geometric center of the boat, and Azi should (but doesn't) account for that. The simulator uses the geometric center, not the real COM. 

Could either of these explain it, @zachgoins ?

closed 2015-05-27 by zachgoins