uf-mil-archive/PropaGator #27: Azi drive produces a torque with opposite sign of desired

fnivek, 2015-06-10:
If a positive (clockwise) torque is given azi drive produces a negative torque (counterclockwise). I have temporarily fixed it with this [commit](https://github.com/uf-mil/PropaGator/commit/685b2510c86c9075a747a548e6dd5eb7ea6cc6c9) but Jake you may want to look into your math and fix it appropriately.

jpanikulam, 2015-06-10:
Hmm, seems to work right on my machine, but I'm not running the latest Propagator. 

Are you using the x-forward y-right convention? Positive torque would turn the boat right, in that case.

forrestv, 2015-06-10:
So... hopefully everyone is using x-forward y-left (z-up), because, you know, that's what everything else uses and is the ROS standard local coordinate system. Positive torque is CCW rotation/turning left, which contradicts what Kevin said.

fnivek, 2015-06-10:
Sorry I was quite tired when I posted this issue last night I meant to say positive is CCW and negative is CW. As for x forward, y left, and z up, everything except Azi drive follows that convention. Jake positive torque should turn the boat CCW or left, and negative torque should turn the boat CW or right.

fnivek, 2015-06-10:
It looks like there is also a sign error in the Y direction, if azi drive really is x forward Y right z down. In your convention, when given a positive Fy it should move right. On my system a wrench with negative Fy moves right and a wrench with positive Fy moves left contrary to the x forward Y right z down convention. (Fortunately for today's test day this is actually beneficial)

jpanikulam, 2015-06-10:
ROS standards vs. nautical standards, does Poseidon supercede OSRF?

I guess not. Will fix it. 

- Jake