uf-mil-archive/PropaGator #42: AZI drive has a bad wrench zone

fnivek, 2015-07-08:
We have noted that when AZI drive is supplied certain wrenches in the 1st quadrant of the base_link frame, AZI drive over corrects and lurches the boat in the forward left direction.

jpanikulam, 2015-07-08:
Is this a controller problem, or an azi drive problem?

- Jake

fnivek, 2015-07-08:
Didn't specify because I don't know but I'm leaning towards azi right now.

jpanikulam, 2015-07-08:
Pushed a fix to the jacob_planner branch. This is quite a substantial change, but will function as a drop-in replacements. The changes that are important are commits 9b3f91e and 810af70, the first two made on July 8th. I did not force a merge because we have diverged quite a bit.

## Items of note

* I am still using the Fossen convention for coordinate axes (Sorry, didn't want to change my unit tests)
* This uses ultra-optimized generated C from [CVXGEN](http://cvxgen.com/). Don't bother reading the source code in azi_cvx except azi_cvxbind.cpp
* This will run 25-50x faster in most situations. In some, up to 100x. This means you can run it at **500 Hz**, as opposed to the old 20 Hz. Fiddle with the maximum angle change to get better convergence properties
* This will **always** converge, and will come up with a solution that is optimal, even when you issue a command that is out of bounds (This resolves the clipping issue)

## Conclusion

I am only doing all of these wacky things because of the rush situation. Changes are almost entirely undocumented because I made them in the last four hours. If you run into an error relating to the C++ bindings, the azi_drive module or anything involving "cvx", call me on my cell. Godspeed!

@fnivek can you confirm that this resolves the issue?

fnivek, 2015-07-08:
As soon as I have some down time I'll check but it's hectic here do it may be awhile